signetlabdei/lorawan #153: Questions about complete network example

Benoflynn, 2024-02-22:
Hey, I've been using you're complete network example to record data such as packet loss rates vs. no of nodes and delays vs. no. of nodes etc. I had a question regarding the randomness of the simulation as i'm getting the same result for packet loss every time when I use the same number of nodes. Also is the placement of the nodes a set location in the 'building' or is this being randomly generated in some kind of way.

I'm also curious about how the traffic is being generated in relation to no. of packets + rate per node, interarrival stats and the overall statistical behaviour of how the traffic is generated.

Any help in regards to this would be great! Thanks.

Benoflynn, 2024-02-23:
Also I was curious if you guys have added a protocol layer such as HTTP or MQTT to this module before ... i'm considering attempting this myself and was just curious as to how complex it might be with the LoRaWAN module.

non-det-alle, 2024-02-28:
> I had a question regarding the randomness of the simulation as I'm getting the same result for packet loss every time when I use the same number of nodes.

[`--RngRun=#`](https://www.nsnam.org/docs/release/3.40/manual/html/random-variables.html#id1)

> Also I was curious if you guys have added a protocol layer such as HTTP or MQTT to this module before

In this [extension](https://github.com/non-det-alle/elora) I developed, where I was able to send HTTP/UDP packets from simulated gateways to a real network server outside the simulation. Otherwise you could take a look at other ns-3 modules for HTTP and MQTT to combine with this one to achieve your goals inside the simulation. 

For all your other questions, please read the code contained in complete-network-example.cc - the answers are written there in the simulation setup.  For further clarifications, I suggest you check out the [ns-3 manual](https://www.nsnam.org/docs/release/3.40/manual/html/index.html) or the [ns-3 tutorial](https://www.nsnam.org/docs/release/3.40/tutorial/html/index.html) to get started with ns-3's [key abstractions](https://www.nsnam.org/docs/release/3.40/tutorial/html/conceptual-overview.html#key-abstractions).

closed 2024-03-28 by non-det-alle