signetlabdei/lorawan #159: Unable to read LoraFrameHeader for uplink messages (using gw mac)

phanijsp, 2024-03-22:
Trying to get address of ED in gateway-lorawan-mac.cc when GW  receives an uplink message. 
Below screenshot shows the changes I have made to the GW mac.
[image]

I have captured the below log before the deserialize method of LoraFrameHeader throws errors and gets stuck in a loop.
 Please note that there is only 1 ED and 1 GW in this simulation. So the packet being serialized and deserialized in the log are the same. 
[image]
 

Screenshot of the error:
[image]

## Specifications

  - ns-3 version: 3.40
  - lorawan module version: 0.3.1
  - Platform: Arch Linux, 6.8.1-arch1-1 
  - Compiler: gcc (GCC) 13.2.1 20230801

non-det-alle, 2024-03-22:
Hello, you get the error because you are trying to remove the frame header without removing the mac header beforehand. As you see in your first screenshot, the mac header is only peeked - not removed - so when you call `RemoveHeader` ns-3 looks for the byte structure of a frame header but finds the bytes of the mac header instead. Try with something like:

```
        Ptr<Packet> secondPacketCopy = packet->Copy();
        LorawanMacHeader tmpMacHdr;
        secondPacketCopy->RemoveHeader(tmpMacHdr);
        LoraFrameHeader frameHdr;
        frameHdr.SetAsUplink();
        secondPacketCopy->RemoveHeader(frameHdr);
```

Here I'm making another copy of the Packet to avoid modifying it if you want to preserve the original one.

phanijsp, 2024-03-22:
Thank you, that worked.

closed 2024-03-22 by phanijsp