signetlabdei/lorawan #162: How to integrate a new model

Vidushi0001, 2024-04-16:
I have worked on drafting a code for classBEndDeviceLorawanMac and an example script to test that, but I need certain instructions on how to integrate it with the module and run it, can someone please guide?
## Specifications

  - ns-3 version:
  - lorawan module version:
  - Platform:
  - Compiler:

non-det-alle, 2024-04-29:
From the top of my head, here are some of the things you will probably need to do:
* Extend `ClassALorawanEndDevice` with your new class B model, and introduce therein the logic for scheduling periodic beacon and ping slots
* Add class B MAC commands by extending the `MacCommand` class
* Modify/extend the `NetworkServer` to manage class B devices
* ...

Please be more specific with your question.

Vidushi0001, 2024-05-04:
@non-det-alle I plan to extend the functionality of this module to classBLorawanEndDevice also, for the same I need to know the steps and places where I need to make changes to integrate the classBLorawanEndDevice class in the module.