signetlabdei/lorawan #165: onoffhelper

Gabriela3000, 2024-05-07:
## Expected Behavior
Hello and thank you very much for all the efforts
Does this new version of lora support ns3 traffic modules such as onoffhelper, udp, tcp, etc.
Best regards

## Actual Behavior

## Steps to Reproduce the Problem

  1.
  2.
  3.

## Specifications

  - ns-3 version:
  - lorawan module version:
  - Platform:
  - Compiler:

non-det-alle, 2024-07-01:
Hi, what do you mean by "support"? This module does not depend on the other ns-3 modules you mention because they are not strictly required to simulate a LoRa/LoRaWAN network. But you can still (I have done it for specific projects in the past) import these other modules in your local copy of the code and add their functionality based on your needs. For example with minor modifications you could write an application for gateways that uses UDP to forward LoRa traffic to the server.

non-det-alle, 2024-07-28:
Closing for lack of activity

closed 2024-07-28 by non-det-alle