signetlabdei/lorawan #167: How to use NetAnim while simulating lora?

SSFG1992, 2024-06-29:
## Steps to Reproduce the Problem

Hello, 
Is it possible to make lora simulation visual with NetAnim? Could you give me some tutorials?

## Specifications

  - ns-3 version: 3.41
  - lorawan module version:
  - Platform: Ubuntu 22.04
  - Compiler:

non-det-alle, 2024-07-01:
Hi, NetAnim support is only partial. You can track the node's positions using ns-3's `AnimationInterface` class, but to the best of my knowledge it does not implement support for LoRa nor tracing generic wireless packets. For more info, please refer to ns-3's [`animation` module documentation](https://www.nsnam.org/docs/release/3.41/models/html/animation.html).

Supporting this would require a pull request to the main ns-3 repository with changes to the `AnimationInterface` class; it could be a nice and easy integration, I'll add this to the enhancement tracker.

non-det-alle, 2024-07-28:
Closing for lack of activity

closed 2024-07-28 by non-det-alle