signetlabdei/lorawan #168: rs-example code

SonileKM, 2024-07-16:
I am running example code for RS LoRa and its giving the following error:

fatal error: ns3/lora-module.h: no such file or directory.
In the code there is a statement ; #include <ns3/lora-module.h>

I tried to check for this header file and associated code but i cant find it in either the ns3.40 folder or the ns-3-Dev folder. I have also tried to check online. Please help me resolve this. 

## Specifications

  - ns-3 version: 3.40
  - lorawan module version: v0.3.1
  - Platform: ubuntu linux 20.04
  - Compiler: G++

non-det-alle, 2024-07-16:
Hello,
What are _rs-example_ and _RS LoRa_? To my knowledge, they are not part of this repository.
That may be where the problematic code comes from.
Let me know.

SonileKM, 2024-07-18:
Hello Alessandro,
First of all thanks for responding. Allow me to also apologise for taking
it for granted that the terms rs-example and RS LoRa are known. I am
studying and following up the following work in the article: "Improving
Reliability and Scalability of
LoRaWANs Through Lightweight Scheduling", where the authors developed the
RS LoRa algorithm (R for Reliability and S for Scalability). I downloaded
the code from the following provided in the article: https://github.com/
networkedsystems/lora-ns
I was trying to run the code in the examples folder to understand how it
works and that's how the error mentioned earlier was coming up. Here is a
sample of the code i was running known as rs-example.cc

#include "ns3/non-communicating-net-device.h"
#include <ns3/okumura-hata-propagation-loss-model.h>
#include <ns3/core-module.h>
#include <ns3/packet.h>
#include <ns3/lora-module.h>
#include <ns3/spectrum-module.h>
#include <ns3/mobility-module.h>
#include <ns3/spectrum-value.h>
#include <ns3/spectrum-analyzer.h>
//#include <ns3/log.h>
//#include <string>
#include <iostream>
#include <ns3/isotropic-antenna-model.h>
#include <ns3/trace-helper.h>
#include <ns3/drop-tail-queue.h>
#include <unordered_map>
#include "ns3/network-module.h"
#include "ns3/csma-module.h"
#include "ns3/internet-module.h"
#include <ns3/gw-trailer.h>
NS_LOG_COMPONENT_DEFINE ("lora");

using namespace ns3;
using namespace std;

/////////////////////////////////
// Configuration
/////////////////////////////////
//std::vector <int, double> marked_list;
double length = 2000;           //!< Square city with length as distance
int pktsize = 51;              //!< size of packets, in bytes
int duration = 60*60;//10000;   //!< Duration of the simulation
double interval = 10;        // interval between packets, minutes
bool verbose = false;          // enable logging (different from trace)
bool nakagami = true;         // enable nakagami path loss
bool dynamic = false;          // enable random moving of pan node
uint32_t nSensors = 10; // numbenir of sent packets
uint32_t nGateways = 1; // numbenir of sent packets
Ptr<OutputStreamWrapper> m_stream = 0; // stream for waterfallcurve
Ptr<UniformRandomVariable> randT = CreateObject<UniformRandomVariable> ();
uint32_t interT = 600; // time between a retransmissions.
std::unordered_map<uint32_t,std::tuple<uint32_t,uint32_t,uint32_t,uint32_t,
uint32_t,uint32_t> > errorMap;
//errormap: transmitted, received, received unique, received original,
xlocation, ylocation
std::unordered_map<uint32_t,Ptr<LoRaNetDevice> > deviceMap;
Mac32Address server;
NetDeviceContainer gateways;
uint8_t offsets [7] = {2,3,3,1,2,2,3};
/////////////////////////////////
// End configuration
/////////////////////////////////

    uint32_t
GetClosestGateway (Ptr<MobilityModel> location)
{
    uint32_t closestNode;
[... 235 lines cut ...]
    m_stream = ascii.CreateFileStream("data7loraTSCH1_100.csv");
    *m_stream->GetStream() << "#Scenario 100 nodes with " << (int)nGateways
<< " gateways and " << (int)nSensors << " nodes on a square field with side
" << length << " meter" <<std::endl;
    for (uint8_t iterationI=0;iterationI<10;iterationI++){
        std::cout << "Iteration: " << (int)iterationI << std::endl;
        mainBody();
        for ( auto it = errorMap.begin(); it !=errorMap.end();++it)
        {
            uint32_t addr = it->first;
            Ptr<LoRaNetDevice> netdevice = deviceMap[addr];
            std::tuple<uint32_t,uint32_t,uint32_t,uint32_t,uint32_t,uint32_t>
tuple = it->second;
            *m_stream->GetStream() << addr << "," << std::get<0>(tuple)<<
"," << std::get<1>(tuple) << "," <<   std::get<2>(tuple) << "," << std::get<
3>(tuple) << "," << std::get<4>(tuple) << "," << std::get<5>(tuple) <<","<<
netdevice->GetAvgRetransmissionCount() << "," << netdevice->GetAvgTime() <<
"," << netdevice->GetMissed() << "," << netdevice->GetArrived() << std
::endl;
        }
        errorMap.clear();
        Simulator::Destroy ();
    }
    return 0;
}

SonileKM, 2024-07-18:
> First of all, please use ns-3.41 as that is the officially supported
> version of ns-3 by the latest version of the lorawan module.
> Regardless, I cannot find the statement mentioned above in the codebase,
> and everything seems to compile without errors by following the guide in
> the README. Which example are you using?

Dear Matteo,
Thanks so much for the advice to use ns-3.41. Here is the code for
rs-example.cc that Iam trying to run.#include
"ns3/non-communicating-net-device.h"
#include <ns3/okumura-hata-propagation-loss-model.h>
#include <ns3/core-module.h>
#include <ns3/packet.h>
#include <ns3/lora-module.h>
#include <ns3/spectrum-module.h>
#include <ns3/mobility-module.h>
#include <ns3/spectrum-value.h>
#include <ns3/spectrum-analyzer.h>
//#include <ns3/log.h>
//#include <string>
#include <iostream>
#include <ns3/isotropic-antenna-model.h>
#include <ns3/trace-helper.h>
#include <ns3/drop-tail-queue.h>
#include <unordered_map>
#include "ns3/network-module.h"
#include "ns3/csma-module.h"
#include "ns3/internet-module.h"
#include <ns3/gw-trailer.h>
NS_LOG_COMPONENT_DEFINE ("lora");

using namespace ns3;
using namespace std;

/////////////////////////////////
// Configuration
/////////////////////////////////
//std::vector <int, double> marked_list;
double length = 2000;           //!< Square city with length as distance
int pktsize = 51;              //!< size of packets, in bytes
int duration = 60*60;//10000;   //!< Duration of the simulation
double interval = 10;        // interval between packets, minutes
bool verbose = false;          // enable logging (different from trace)
bool nakagami = true;         // enable nakagami path loss
bool dynamic = false;          // enable random moving of pan node
uint32_t nSensors = 10; // numbenir of sent packets
uint32_t nGateways = 1; // numbenir of sent packets
Ptr<OutputStreamWrapper> m_stream = 0; // stream for waterfallcurve
Ptr<UniformRandomVariable> randT = CreateObject<UniformRandomVariable> ();
uint32_t interT = 600; // time between a retransmissions.
std::unordered_map<uint32_t,std::tuple<uint32_t,uint32_t,uint32_t,uint32_t,
uint32_t,uint32_t> > errorMap;
//errormap: transmitted, received, received unique, received original,
xlocation, ylocation
std::unordered_map<uint32_t,Ptr<LoRaNetDevice> > deviceMap;
Mac32Address server;
NetDeviceContainer gateways;
uint8_t offsets [7] = {2,3,3,1,2,2,3};
/////////////////////////////////
// End configuration
/////////////////////////////////

    uint32_t
GetClosestGateway (Ptr<MobilityModel> location)
{
    uint32_t closestNode;
    double smallestDistance = 99999999;
    for (uint32_t i = 0; i<gateways.GetN (); i++)
    {
        double distanceToI = location->GetDistanceFrom (gateways.Get(i)->
GetNode ()->GetObject<MobilityModel> ());
        if (distanceToI < smallestDistance)
        {
            smallestDistance = distanceToI;
            closestNode = gateways.Get(i)->GetNode ()->GetId();
        }
[... 225 lines cut ...]
    m_stream = ascii.CreateFileStream("data7loraTSCH1_100.csv");
    *m_stream->GetStream() << "#Scenario 100 nodes with " << (int)nGateways
<< " gateways and " << (int)nSensors << " nodes on a square field with side
" << length << " meter" <<std::endl;
    for (uint8_t iterationI=0;iterationI<10;iterationI++){
        std::cout << "Iteration: " << (int)iterationI << std::endl;
        mainBody();
        for ( auto it = errorMap.begin(); it !=errorMap.end();++it)
        {
            uint32_t addr = it->first;
            Ptr<LoRaNetDevice> netdevice = deviceMap[addr];
            std::tuple<uint32_t,uint32_t,uint32_t,uint32_t,uint32_t,uint32_t>
tuple = it->second;
            *m_stream->GetStream() << addr << "," << std::get<0>(tuple)<<
"," << std::get<1>(tuple) << "," <<   std::get<2>(tuple) << "," << std::get<
3>(tuple) << "," << std::get<4>(tuple) << "," << std::get<5>(tuple) <<","<<
netdevice->GetAvgRetransmissionCount() << "," << netdevice->GetAvgTime() <<
"," << netdevice->GetMissed() << "," << netdevice->GetArrived() << std
::endl;
        }
        errorMap.clear();
        Simulator::Destroy ();
    }
    return 0;
}

non-det-alle, 2024-07-18:
Hi @SonileKM, 

Your questions are about a different codebase, please open an issue on the [correct repository](https://github.com/networkedsystems/lora-ns3).
We have no knowledge or affiliation with the code you are referring to.

closed 2024-07-18 by non-det-alle

SonileKM, 2024-07-19:
Thanks so much Alessandro. Will do as advised.