signetlabdei/lorawan #170: Challenges understanding error in NS3 lorawan module

SonileKM, 2024-08-07:
## Expected Behavior: Expected lora code to run since the lora module has been installed and there are no longererror for the header files

## Actual Behavior
This is the error we are getting: The error shown in the screen shot attached and we can' seem to figure out what the real issue is and to get around it.

## Steps to Reproduce the Problem  
  1.
  2.
  3.

## Specifications

  - ns-3 version: 3.40
  - lorawan module version: v0.3.1
  - Platform: Ubuntu linux
  - Compiler: gcc
[image]

non-det-alle, 2024-08-08:
Hello, release v0.3.1 of the lorawan module depends on ns-3.41. You are using ns-3.40, that's what is causing the errors.

SonileKM, 2024-08-09:
Thanks for your response Alessandro. We actually installed ns-341 and it is
giving us an error that we are failing to resolve. The compiler is only
recognising header files in the lorawan module but not those that are in
the folder for the source code we are trying to run. Here is a snapshot of
the output we are getting if we run it from the ns-3.41 directory or from
the ns-3-dev directory.

non-det-alle, 2024-08-12:
> Here is a snapshot of the output we are getting if we run it from the ns-3.41 directory or from the ns-3-dev directory.

Hi @SonileKM, I cannot find the snapshot you mention.

SonileKM, 2024-08-19:
Please see the attachment for the screenshot. Please advise if you are able
to view it.

SonileKM, 2024-08-24:
Following your advice, here it the attachment. 
[image]

non-det-alle, 2024-08-26:
> The compiler is only recognising header files in the lorawan module but not those that are in the folder for the source code we are trying to run.

Hi @SonileKM, where did you place `gw-trailer.h` and the other files you want to include in `rs-example.cc`? This is a bit out of scope for us to help you, given that the files of this repo are not the source of your errors. 

If you need to run a simulation with additional libraries, maybe refer to the multi-file examples provided by ns-3 in the [scratch directory](https://gitlab.com/nsnam/ns-3-dev/-/tree/ns-3.41/scratch). If you want to create a new ns-3 module with the files you need to import, checkout the ns-3 manual on [how to create a new module](https://www.nsnam.org/docs/manual/singlehtml/index.html#adding-a-new-module-to-ns3) instead. Finally, if you want to  integrate your additional files in a copy of this repository, I suggest you start studying both [the model](https://signetlabdei.github.io/lorawan/models/build/html/lorawan.html) and the code directly to understand where you could fit the additional functionality.

SonileKM, 2024-08-26:
Thank you so much Alessandro. I will definitely follow through with the
suggestions you have given.

Regards,
Sonile

SonileKM, 2024-08-26:
I place gw-trailer and other header files in a model folder which is inside
the project folder. The rs-example code is also in the example folder
inside the project folder.

non-det-alle, 2024-08-26:
Ok, so if you are putting your library files (`gw-trailer.h`, `gw-trailer.cc`, etc.) in the [model](/signetlabdei/lorawan/blob/develop/model) directory of this repo, you should also add them in [CMakeLists.txt](/signetlabdei/lorawan/blob/develop/CMakeLists.txt) to allow the ns-3 build system to pick them up at compilation time. Make sure to put everything inside the `ns3` namespace, otherwise include directories as `#include "ns3/gw-trailer.h"` won't work.

If that does not produce compilation errors, try running `rs-example.cc` as you did before. It's ok to leave it in the `scratch` folder (the main difference is that in `scratch` files are automatically added to the build system without having to specify them in a `CMakeLists.txt` file, as you would need to do instead in the `examples` directory).

SonileKM, 2024-08-26:
Thank you so much Alessandro. I will try this.

Regards,
Sonile

SonileKM, 2024-09-06:
Dear Alessandro,
Thank you so much for this valuable advice. The error due to external
library files (gw-trailer.h, ge-trailer.cc) has been resolved. I will
definitely get back to you if i encounter other challenges.

closed 2024-09-06 by non-det-alle