signetlabdei/lorawan #171: ED-side Adaptive Data Rate (ADR) update and ADR Backoff

geraldosarmento, 2024-09-12:
Hi,

Upon reviewing the ClassAEndDeviceLorawanMac::SendToPhy method, I noticed that it implements an ED-side ADR version from an outdated LoRaWAN specification. Are there any plans to update it to version 1.1, including the [ADR Backoff feature](https://learn.semtech.com/mod/book/view.php?id=174&chapterid=162#:~:text=Adaptive%20Data%20Rate%20(ADR)%20describes,the%20number%20of%20retransmissions%20made.)?

https://github.com/signetlabdei/lorawan/blob/44c01e79eb44869b50039ecd6a66380f66b5415f/model/class-a-end-device-lorawan-mac.cc#L88C5-L94C6

Looking forward to your answer, thanks!

## Specifications

  - ns-3 version: 3.42
  - lorawan module version: develop
  - Platform: Linux
  - Compiler: g++

non-det-alle, 2024-09-12:
Hi @geraldosarmento, yes we are aware. You can find it among other ongoing changes in issue #139. I already have an up-to-date implementation of the ADR backoff procedure in a fork, it's a matter of finding the time to bring it over. I'll leave this open as a reminder for me to start from this when I find the time.

non-det-alle, 2025-11-09:
ADR from latest specification (1.0.4) has been implemented following #198

closed 2025-11-09 by non-det-alle