signetlabdei/lorawan #189: static assertion failed: Use CreateObject() instead of Create()

andreicezar, 2025-06-14:
Hi, 
I used this commands to take and build the repo:
```
cd ~/ns-3/src
git clone https://github.com/signetlabdei/lorawan.git
cd ~/ns-3
mkdir cmake-cache
cmake -S . -B cmake-cache -DCMAKE_BUILD_TYPE=Release -DNS3_EXAMPLES=ON -DNS3_TESTS=OFF -DNS3_ENABLED_MODULES=lorawan
cmake --build cmake-cache -j$(nproc)
```

I got this compiling error:
```
In file included from /home/andrei/ns-3/src/core/model/attribute.h:11,
                 from /home/andrei/ns-3/src/core/model/attribute-accessor-helper.h:11,
                 from /home/andrei/ns-3/src/core/model/attribute-helper.h:12,
                 from /home/andrei/ns-3/build/include/ns3/attribute-helper.h:1,
                 from /home/andrei/ns-3/src/network/model/address.h:14,
                 from /home/andrei/ns-3/build/include/ns3/address.h:1,
                 from /home/andrei/ns-3/src/lorawan/model/lora-device-address.h:12,
                 from /home/andrei/ns-3/src/lorawan/model/end-device-lorawan-mac.h:15,
                 from /home/andrei/ns-3/src/lorawan/model/end-device-lorawan-mac.cc:12:
/home/andrei/ns-3/src/core/model/ptr.h: In instantiation of ‘ns3::Ptr<T> ns3::Create(Ts&& ...) [with T = ns3::lorawan::LinkAdrAns; Ts = {bool&, bool&, bool&}]’:
/home/andrei/ns-3/src/lorawan/model/end-device-lorawan-mac.cc:818:53:   required from here
/home/andrei/ns-3/src/core/model/ptr.h:441:19: error: static assertion failed: Use CreateObject() instead of Create() for Object subclasses
  441 |     static_assert(!std::is_base_of_v<Object, T>,
      |                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/andrei/ns-3/src/core/model/ptr.h:441:19: note: ‘!(bool)std::is_base_of_v<ns3::Object, ns3::lorawan::LinkAdrAns>’ evaluates to false
/home/andrei/ns-3/src/core/model/ptr.h: In instantiation of ‘ns3::Ptr<T> ns3::Create(Ts&& ...) [with T = ns3::lorawan::DutyCycleAns; Ts = {}]’:
/home/andrei/ns-3/src/lorawan/model/end-device-lorawan-mac.cc:828:55:   required from here
/home/andrei/ns-3/src/core/model/ptr.h:441:19: error: static assertion failed: Use CreateObject() instead of Create() for Object subclasses
/home/andrei/ns-3/src/core/model/ptr.h:441:19: note: ‘!(bool)std::is_base_of_v<ns3::Object, ns3::lorawan::DutyCycleAns>’ evaluates to false
/home/andrei/ns-3/src/core/model/ptr.h: In instantiation of ‘ns3::Ptr<T> ns3::Create(Ts&& ...) [with T = ns3::lorawan::DevStatusAns; Ts = {unsigned char&, unsigned char&}]’:
/home/andrei/ns-3/src/lorawan/model/end-device-lorawan-mac.cc:858:55:   required from here
/home/andrei/ns-3/src/core/model/ptr.h:441:19: error: static assertion failed: Use CreateObject() instead of Create() for Object subclasses
/home/andrei/ns-3/src/core/model/ptr.h:441:19: note: ‘!(bool)std::is_base_of_v<ns3::Object, ns3::lorawan::DevStatusAns>’ evaluates to false
/home/andrei/ns-3/src/core/model/ptr.h: In instantiation of ‘ns3::Ptr<T> ns3::Create(Ts&& ...) [with T = ns3::lorawan::NewChannelAns; Ts = {bool&, bool&}]’:
/home/andrei/ns-3/src/lorawan/model/end-device-lorawan-mac.cc:923:56:   required from here
/home/andrei/ns-3/src/core/model/ptr.h:441:19: error: static assertion failed: Use CreateObject() instead of Create() for Object subclasses
/home/andrei/ns-3/src/core/model/ptr.h:441:19: note: ‘!(bool)std::is_base_of_v<ns3::Object, ns3::lorawan::NewChannelAns’ evaluates to false
[ 85%] Building CXX object src/lorawan/CMakeFiles/lorawan.dir/model/class-a-end-device-lorawan-mac.cc.o
[ 86%] Building CXX object src/lorawan/CMakeFiles/lorawan.dir/model/gateway-lora-phy.cc.o
[ 86%] Building CXX object src/lorawan/CMakeFiles/lorawan.dir/model/end-device-lora-phy.cc.o
[ 86%] Building CXX object src/lorawan/CMakeFiles/lorawan.dir/model/simple-end-device-lora-phy.cc.o
gmake[2]: *** [src/lorawan/CMakeFiles/lorawan.dir/build.make:206: src/lorawan/CMakeFiles/lorawan.dir/model/end-device-lorawan-mac.cc.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
In file included from /home/andrei/ns-3/src/core/model/attribute.h:11,
                 from /home/andrei/ns-3/src/core/model/attribute-accessor-helper.h:11,
                 from /home/andrei/ns-3/src/core/model/attribute-helper.h:12,
                 from /home/andrei/ns-3/build/include/ns3/attribute-helper.h:1,
                 from /home/andrei/ns-3/src/network/model/address.h:14,
                 from /home/andrei/ns-3/build/include/ns3/address.h:1,
                 from /home/andrei/ns-3/src/lorawan/model/lora-device-address.h:12,
                 from /home/andrei/ns-3/src/lorawan/model/end-device-lorawan-mac.h:15,
                 from /home/andrei/ns-3/src/lorawan/model/class-a-end-device-lorawan-mac.h:15,
                 from /home/andrei/ns-3/src/lorawan/model/class-a-end-device-lorawan-mac.cc:13:
/home/andrei/ns-3/src/core/model/ptr.h: In instantiation of ‘ns3::Ptr<T> ns3::Create(Ts&& ...) [with T = ns3::lorawan::RxParamSetupAns; Ts = {bool&, bool&, bool&}]’:
/home/andrei/ns-3/src/lorawan/model/class-a-end-device-lorawan-mac.cc:570:32:   required from here
/home/andrei/ns-3/src/core/model/ptr.h:441:19: error: static assertion failed: Use CreateObject() instead of Create() for Object subclasses
  441 |     static_assert(!std::is_base_of_v<Object, T>,
      |                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/andrei/ns-3/src/core/model/ptr.h:441:19: note: ‘!(bool)std::is_base_of_v<ns3::Object, ns3::lorawan::RxParamSetupAns>’ evaluates to false
gmake[2]: *** [src/lorawan/CMakeFiles/lorawan.dir/build.make:222: src/lorawan/CMakeFiles/lorawan.dir/model/class-a-end-device-lorawan-mac.cc.o] Error 1
[ 87%] Linking CXX executable ../../../../build/src/propagation/examples/ns3-dev-jakes-propagation-model-example
[ 87%] Linking CXX executable ../../../../build/src/buildings/examples/ns3-dev-outdoor-random-walk-example
[ 87%] Linking CXX executable ../../../../build/src/buildings/examples/ns3-dev-outdoor-group-mobility-example
gmake[1]: *** [CMakeFiles/Makefile2:1617: src/lorawan/CMakeFiles/lorawan.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:136: all] Error 2
```

andreicezar, 2025-06-14:
Hi,
There are 3 versions of the issue I have created.
There are compilation errors that appears when I am doing the steps listed in the issue created.
The generated text is used for making it easier to read and easier to identify the issue, as well as the versions used.
If you consider it’s important to identific these compilation errors, you can take it into consideration.
As for now I will try to solve them whenever they appear. 
Have a nice day.

non-det-alle, 2025-06-15:
Hi, yeah sorry. I think maybe I should have been less strict and not close the issue for just using generated text. I am reopening it.

But let me first elaborate. In general, my opinion is that generated text is often less clear and more sloppy than manually describing the specific steps/commands you run to get your output. It fills everything with generic text and it is not up to the point, making us maintainers waste precious time. We make an effort to investigate your errors and share our expertise to help, so I think it's fair that submissions are also of some effort. On this matter, you can find [here](https://curl.se/dev/contribute.html#on-ai-use-in-curl) a much more authoritative opinion than mine.

In order to help you, I need to know a couple of things not included in your issue. Which version of the LoRaWAN module are you working with? You say the latest but these things may change with time. Which commit are you on?

I feel that your problem could be related to a mismatch in the ns-3 version you are running this module with. Your issue describes unusual installation and configuration steps. Follow our README.md file to find how to install both the module and ns-3 to ensure their version compatibility.

closed 2025-07-21 by andreicezar